fetchrobotics/fetch_ros #55: Can not launch the simulation for fetch robot. Segmentation fault(core dumped)

josephinemonica, 2016-09-22:
I tried to launch the simulation 

roslaunch fetch_gazebo simulation.launch

but it failed with the following error message:

```
[urdf_spawner-5] process has finished cleanly
log file: /home/osboxes/.ros/log/2599ce20-806c-11e6-b92a-000c29a1f7dd/urdf_spawner-5*.log
[ WARN] [1474511279.383347311, 0.309000000]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1474511279.403651504, 0.309000000]: Started arm_controller/gravity_compensation
[ INFO] [1474511279.819152703, 0.309000000]: Started base_controller
[ WARN] [1474511280.117777683, 0.309000000]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1474511280.546230380, 0.309000000]: Started bellows_controller
[ INFO] [1474511280.572457414, 0.309000000]: Finished initializing FetchGazeboPlugin
[ INFO] [1474511280.710423220, 0.377000000]: Started gripper_controller/gripper_action
[ INFO] [1474511280.711726394, 0.377000000]: Started head_controller/follow_joint_trajectory
[ INFO] [1474511280.712074683, 0.377000000]: Started arm_controller/follow_joint_trajectory
Segmentation fault (core dumped)
[gazebo_gui-3] process has died [pid 2565, exit code 139, cmd /opt/ros/indigo/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/osboxes/.ros/log/2599ce20-806c-11e6-b92a-000c29a1f7dd/gazebo_gui-3.log].
log file: /home/osboxes/.ros/log/2599ce20-806c-11e6-b92a-000c29a1f7dd/gazebo_gui-3*.log
[prepare_robot-6] process has finished cleanly
log file: /home/osboxes/.ros/log/2599ce20-806c-11e6-b92a-000c29a1f7dd/prepare_robot-6*.log

```
I notice it's written segmentation fault.

I have no problem with gazebo, as I can launch gazebo simulation using other robots.
In fact if I change the robot from fetch to freight. It can launch!
`roslaunch fetch_gazebo simulation.launch robot:="freight"`

So I think the problem is in the fetch robot itself.

mikeferguson, 2016-09-22:
@josephinemonica I've not had any other reports like this, and I've just locally tested and cannot reproduce. Can you give us a bit more info about your install?

 * What version of ubuntu? 14.04, 16.04, something different?
 * Is everything installed from debians? If not, please let us know which repos are being built from source (and which branches).
 * If everything is from debians, can you make sure that all the versions are up to date (ROS doesn't really guarantee all that much in terms of ABI compatibility, so you really need to have all the packages up to date)

josephinemonica, 2016-09-22:
Hi, thanks for your response.

I use : Ubuntu 14.04, ROS indigo, gazebo 2.
I installed using sudo apt-get install ros-indigo-fetch*
I tried to launch roslaunch fetch_gazebo simulation.launch, and other launch files , but they all failed. If I change the robot to freight, they can work.

josephinemonica, 2016-09-23:
Sometimes the error produced is:
VMware: vmw_ioctl_command error Cannot allocate memory.
Aborted (core dumped)

I have tried to allocate more memory too.

josephinemonica, 2016-09-23:
@mikeferguson How may bytes of RAM is needed to run fetch simulation? I have allocated more memory. The error switches between "can not allocate memory" and "sgmentation fault".

mikeferguson, 2016-10-26:
Looking at the trace, it looks "gazebo_gui-3" is the process that died? It seems this really is a Gazebo in VMware problem, since I haven't seen any other reports of this being an issue with the fetch_gazebo packages.

I typically don't run ROS in virtual machines, but it looks like there are some other people who have run into similar errors: http://answers.gazebosim.org/question/13214/virtual-machine-not-launching-gazebo/ I'd suggest trying that, and if that does not work, seeing if the Gazebo folks have any ideas by asking on answers.gazebosim.org

closed 2016-10-26 by mikeferguson

josephinemonica, 2016-10-26:
I opened the link and tried the suggestion there: unchecking "Accelerate 3D Graphics" option in the virtual machine settings

And now I can open the fetch robot simulation!
Thanks @mikeferguson for the pointer

I really don't know why, cause I can open other robots (including freight robot) in gazebo simulation, except for this fetch..
Maybe this fetch robot is quite heavy in memory etc.

Anw It's working now, even though lags a lot. :)