fetchrobotics/fetch_ros #56: Support for ROS Kinetic

ruffsl, 2016-10-26:
I have a Ubuntu 16.04 workstation and was trying to play around with the fetch_gazebo_demo package, and so was atempting build fetch_navigation package, but happened to hit what I think is some issues with recent changes in MoveIt!, see error output bellow. Any ETA on support on Kinetic for fetch_ros or fetch robots ingeneral?

http://pastebin.com/ARpmDGtA

mikeferguson, 2016-10-26:
We actually had all of the simulation working in late August during World MoveIt Day, but MoveIt in Kinetic has been a somewhat moving target as of late. It actually looks like only the ikfast is failing to build -- I think we'll be able to fix that relatively quickly.

As for Kinetic on actual robots, that's a bit further off -- it actually requires also getting 16.04 support finished. Is this something your lab is looking for?

mikeferguson, 2016-10-26:
@ruffsl -- I don't have a kinetic machine here to test with tonight, but I think https://github.com/fetchrobotics/fetch_ros/pull/57 will fix the build on Kinetic since this was just the new URDF lib version issue

mikeferguson, 2016-10-26:
And about as soon as I pressed enter, I realized that might not fix everything. We probably also need to add C11 to the cmake, I'll try to test that on a 16.04 machine tomorrow.

ruffsl, 2016-10-26:
Thanks @mikeferguson, if you need someone to check the PR, I'd be happy to test using it.

> As for Kinetic on actual robots, that's a bit further off -- it actually requires also getting 16.04 support finished. Is this something your lab is looking for?

My lab setup our workstations with 16.04 for the current ROS LTS before our fetchs arrived, realizing that they had been shipped with 14.04. Its mainly a desire to have our dev environments match our targets', but also the issue in rolling back would be troublesome, as the kernel shipped with 14.04 might be tedious to get booting on latest gen hardware (like my new laptop or PCIe M2/U2 enabled motherboard).

Being able to at least build and iterate locally on kinetic would be nice, as for now we're just working above of what's already here.

mikeferguson, 2016-10-27:
@ruffsl -- moving your desktops to 14.04 might not be as hard as you expect -- Ubuntu has done a much better job lately in having upgraded installers and the 14.04.5 installer includes a 4.4 kernel and latest graphics support. We've not experienced much issue using these newer installers even with brand new laptops.

mikeferguson, 2016-10-27:
Build errors should now be fixed on the mainline, I'm going to leave this open for further discussion on Kinetic on real robots.

pholthau, 2018-05-18:
I currently use kinetic on my local machine connected to a robot that still operates on indigo. So far without problems - everything builds on kinetic and seems to work as intended.

Still, are there any plans on updating the robots (maybe directly to melodic)?

moriarty, 2018-07-27:
@ruffsl & @rctoris can we close this?

We're going to release for **18.04** & **Melodic** soon.

On the ros-software side: _most_ things are working and tested on **18.04**. I've updated #63 with some more details.
On the hardware side, I've got a little bit of work to do with the **ps3joy** tweaking, the current work around isn't ready for release, and to port some things from **upstart** to **systemd**.

Unofficially I have **16.04** & **Kinetic** on my personal laptop and haven't had issues (sim only).
But for the actual Fetch hardware, we'll jump to **18.04**.

ruffsl, 2018-07-27:
> We're going to release for 18.04 & Melodic soon.

:tada: ping @BillWSY

knorth55, 2018-11-22:
I still  want kinetic support and deb package release of `fetch_description`.

varunagrawal, 2019-01-25:
@ruffsl @mikeferguson any updates on this?

ruffsl, 2019-01-25:
@varunagrawal , not that I'm aware, I haven't been keeping up with this migration.  
https://github.com/fetchrobotics/fetch_ros/issues/63 seems to show the latest status.

mikeferguson, 2019-01-27:
Just FYI, I left Fetch at the end of 2017, so I can't provide any updates on this.

cjds, 2019-02-13:
@ruffsl @varunagrawal @knorth55  we have been supporting melodic for Fetch and are going to do a deb package for that Target. 

We can probably get a Kinetic one out if people really need it but will melodic work for you or do you speicifically need Kinetic?

@moriarty FYI

moriarty, 2019-02-13:
Closing this issue.

We've missed the boat :ship: on Kinetic. We'll support Indigo and Melodic.

Kinetic, won't be getting Debian releases.

For simulation it should be possible to run Kinetic but on the real hardware we won't be testing the drivers on Kinetic. We have hardware requirements that require running the newer kernel and supporting and testing all versions is problematic.

https://github.com/fetchrobotics/fetch_ros/issues/63

closed 2019-02-13 by moriarty