fetchrobotics/fetch_ros #63: Support of Ubuntu 16.04

panshuyrr, 2017-08-18:
I am working on a project using ROS and Fetch robot - including both working with the real robot as well as the simulation environment.

I am going to install Ubuntu in a new computer. Can I confirm with you whether all the features are fully supported in Ubuntu 16.04.3 LTS? Or Ubuntu 14.04 is the only choice for my new computer?

Thank you very much.

hzyjerry, 2018-01-29:
Hi, I have the same issue. Wondering if there's an updated answer to this at the moment?

Thanks

rctoris, 2018-01-29:
Currently the Fetch is only supported on Ubuntu 14.04. We realize we are nearing the end of support for Ubuntu 14.04 and have started initial investigations and plans for upgrading the research robots to a further version of Ubuntu. At this time we don't have an estimate of when this work will be complete but will update the community as we have additional information.

DanielTakeshi, 2018-05-02:
@rctoris Hi, just wanted to kindly ask if there were any plans to upgrade to Ubuntu 16.04. Thanks!

moriarty, 2018-07-10:
Just as an update: @rctoris mentioned “this summer” ... some preliminary work has started on this.

It was possible to upgrade to 18.04 and Melodic, not all dependencies were available in Melodic, so some were cloned into a catchin workspace, patched and built from source.

There are some issues which we’ll work out before officially releasing... e.g. the default ps3 joypad and latest kernel don’t get along.

But, Fetch was running Melodic, moving & MoveIt! Running today.

I just wanted to update this thread.

DanielTakeshi, 2018-07-10:
@moriarty Thank you! It would be fantastic to have it upgraded to 18.04, and we would upgrade our software in a heartbeat.

moriarty, 2018-07-10:
For anyone with a new laptop running 16.04 (myself personal laptop included) it is possible to clone the repositories and build them with Kinetic in a catkin workspace.

This was covered in another issue. #56, Specifically three comments.

A few changes were required but they were straight forward from the compile errors to fix.

C++11 flags were needed in a few places: https://github.com/fetchrobotics/fetch_ros/issues/56#issuecomment-256253819

build errors were fixed... on Oct 2016: https://github.com/fetchrobotics/fetch_ros/issues/56#issuecomment-256701063
Note: they might be broken again if you take master branches due to some upstream changes. I believe a tf2 back port added a c++11 requirement again.

And @pholthau recently said that running 16.04 locally & connecting to a 14.04 robot was working: https://github.com/fetchrobotics/fetch_ros/issues/56#issuecomment-390138822

But the current focus is on **18.04**

moriarty, 2018-07-18:
Progress Report:
I've opened a PR for Navigation: https://github.com/ros-planning/navigation/pull/755

https://github.com/ros-planning/navigation/issues/749#issuecomment-405812657 which once merged we will get a melodic release for ROS navigation, one of our dependencies :heavy_check_mark: 
https://github.com/ros-planning/navigation/milestone/4

Once that is released some of our smaller dependencies are also ready but not released.

The PR to Navigation breaks **fetch_depth_layer** but I have changes ready to port that to **tf2**.

The PS3Joy pad is a weird. But from @chadrockey & @velveteenrobot & @cjds seemed to know that shaking the joypad (to keep the accelerometer publishing) was a sort-of work around... And that I need to tune some parameters as a proper work around.

warp1337, 2018-08-09:
Hi everyone! 

first things first ... hi Patrick @pholthau I knew our paths would cross at some point ... again ... :D

However, I just built the packages required for the navigation demo (Gazebo7) from source on 16.04. Seems to work as @moriarty pointed out earlier.

My incentive was pretty much that I just wanted to have a look at the "Fetch stack", thus no hardware involved.

 I believe that's also the case for many other ppl who are "still" on 16.04 and therefore on Kinetic. From my perspective it would be great to have "kinetic-experimental" branches and _some_ docs. The docs should clearly state that Kinetic is not supported in production environments but if you want to try some toy examples and get a feeling for the Fetch stuff ... Well why not?

I am willing to document and contribute, if desired.

nickswalker, 2018-10-01:
Is there a preliminary release of `fetch-drivers` available for folks who want to try Melodic on their robots?

moriarty, 2018-10-30:
Sorry for the delay, no ```fetch-drivers``` and the new **upstart->systemd** scripts are two remaining packages which need to be made available, if one were to try a DIY upgrade: which we do not recommend or support.

But we will be testing the packages and upgrade process before the official "upgrade" process is announced and made generally available.

We're working towards making the upgrade process as seamless, stable and well supported for all customers.

moriarty, 2019-02-13:
https://github.com/ros/rosdistro/pull/20301

pholthau, 2019-02-14:
Is there any documented way of upgrading the robot itself?

erelson, 2019-02-14:
@pholthau that documentation is ready in a PR, but pending testing once the 18.04 debians are actually created. Soon :)

If you want to take a look (feedback/questions very welcome), see the PR here: https://github.com/fetchrobotics/docs/pull/60

moriarty, 2019-03-12:
@pholthau fetchrobotics/docs#60 has been merged.

http://docs.fetchrobotics.com/indigo_to_melodic.html

pholthau, 2019-03-12:
@moriarty thanks a lot for this. Will try it out as soon as I can get my hands on the robot again!

moriarty, 2019-03-14:
Warning: 18.04 has been released and works on the physical robot...
**But** we're having trouble identifying and fixing some bugs with the robot in **Gazebo9**.

These issues are being tracked here:
https://github.com/fetchrobotics/fetch_gazebo/issues/37

https://github.com/fetchrobotics/fetch_gazebo/issues/36#issuecomment-463694403 from @nickswalker 
> Yes, navigation has been working fine on the real robot

We'll email everyone who owns a Fetch and announce the 18.04 upgrade after we've solved the Gazebo9 issues.

moriarty, 2019-03-31:
TL;DR:
---------

- We've released **fetch_ros** v0.8.0 with support for **18.04** on Fetch Robots, the process is currently a **DIY for early adopters**.
- @erelson has published http://docs.fetchrobotics.com/indigo_to_melodic.html
- The source code for the site is here: https://github.com/fetchrobotics/docs/ if there are any steps which are unclear please open an issue.
- As early adopters move to 18.04 we'll fix any new issues and release patches
- I'll prepare v0.8.1 as soon as I have more time to test https://github.com/fetchrobotics/fetch_ros/pull/107 on hardware 

Gazebo9 Issues
---------------------

There are still some issues with Gazebo9. But it is much better than it first did and we couldn't be blocked by these issues any longer. The main new changes to make things work:
1. I added "castor wheels" to the gazebo model: https://github.com/fetchrobotics/fetch_gazebo/pull/49
2. @RDaneelOlivav "improved the wheel friction values": https://github.com/fetchrobotics/fetch_gazebo/pull/59

I've bloomed "0.9.0" of fetch_gazebo as it now is, so that at least the ros build farm would test pull requests. We'll still actively work on improving the situation.

If Gazebo9 issues are a blocker
----------------------------------------
For simulation only, I've **released fetch_ros**  into **16.04** and Kinetic - this was not originally planned, and **we won't support 16.04 and Kinetic on the real robot** because we haven't tested the drivers, hardware and firmware on 16.04...

I needed to do a lot of back and forth comparing while looking into the Gazebo9 vs Gazebo7 for debugging an issue. The version of **fetch_ros** released into kinetic is the same version released into indigo. I did all the testing on an 18.04 machine, using a 16.04 docker image, from this Dockerfile: https://github.com/fetchrobotics/fetch_gazebo/issues/46

I hope, that by releasing simulation only support for **16.04**  would help end users, and move everyone forward off of **14.04** which is about to reach **EOL** towards **18.04**

ryom314, 2019-04-05:
Just tried upgrading to *18.04*.
The gripper was working fine before the upgrade, but after the upgrade,

```
ping 10.42.42.43
```

fails and `arp -a` yields

```
? (10.42.42.43) at <incomplete> on eth1
```

Since the step for `ping`ing `10.42.42.43` was in the update procedure I suspected that this might be an issue related to the upgrade.
I'd appreciate any ideas on fixing the issue.

erelson, 2019-04-06:
@ryom314 Sorry to hear about the issue. If you can open a support ticket at support.fetchrobotics.com, we can get started troubleshooting the issue. Thanks!

erelson, 2019-05-10:
@ryom314 I don't believe I've seen any support ticket from you, but I also do not know which robot/research group/university you are working with. Were you able to resolve your issue? Thanks!

ryom314, 2019-05-10:
Thanks for asking. I'm contacting Mark Onaga. A downgrade to 14.04 didn't
resolve the issue so it probably wasn't an OS compatibility issue.

nickswalker, 2019-05-31:
I've just verified that our gripper is unresponsive after the upgrade as well. We'll get a support ticket in soon!

sergeant-wizard, 2019-05-31:
FYI we ended up replacing the MCB for the gripper

sergeant-wizard, 2019-05-31:
(Don't know if the issue was triggered by the update to 18.04 or not.)

erelson, 2019-06-01:
Hi @nickswalker, We'll keep an eye out for the support ticket. Thanks!

@ryom314 @sergeant-wizard glad you guys are back in action. We were able to use your problem board to identify a workaround, should the same thing happen going forward.

moriarty, 2019-06-13:
Just to update the public, the issue from @nickswalker at University of Washington was fixed without replacing the MCB.

@erelson asked in the support ticket:

> Could you report back with the output of `systemctl status robot` for your Fetch?  A good output will include green text indicateing the service is enabled and running.

And then to run:

```
sudo service roscore stop
sudo service roscore start
sudo service robot start
```

moriarty, 2019-06-13:
@erelson should we close this ticket?

I think we can deal with any of the future items under a new issue.

- PS3 Joystick
- Gazebo behaviours
- any other things that come up

moriarty, 2019-06-13:
Confirmed with @erelson over Slack.

I'm closing this. Please see the upgrade guide:

http://docs.fetchrobotics.com/indigo_to_melodic.html

And open any new issues, or support tickets if you have any problems.

closed 2019-06-13 by moriarty