fetchrobotics/fetch_ros #70: /imu has an empty frame_id

BillWSY, 2018-03-02:
in the latest robot_driver package.

mikeferguson, 2018-03-02:
@dbking77

dbking77, 2018-03-05:
@BillWSY What version of robot_drivers package? 
```
cat /opt/ros/indigo/share/package.xml
```

BillWSY, 2018-03-06:
```
cat /opt/ros/indigo/share/fetch_drivers/package.xml

  <name>fetch_drivers</name>
  <version>0.7.28</version>
  <description>
    This is both the ROS drivers and the firmware for Fetch Robotics robots.
  </description>
  <author>Michael Ferguson</author>
  <author>Derek King</author>
  <maintainer email="mferguson@fetchrobotics.com">Michael Ferguson</maintainer>
  <license>Proprietary</license>
```

dbking77, 2018-03-22:
This should fixed in 0.7.29.   @erelson @bluryi are there any plans to sync a new fetch_drivers release to the public repo?

ayzaan, 2018-07-24:
When is the fix going to be released? It looks like the latest available is 0.7.28

moriarty, 2018-07-27:
@rctoris: This sounds like it also effects **14.04.5** and **Indigo** but I've added it to the **18.04** milestone to keep track of it.

moriarty, 2019-03-15:
**0.7.29** was never released for **Indigo**, and at the time we changed how everything internally was built and released (docker, automated nightly builds, different versioning).

@cjds & @bluryi we should discuss if, and how to release this package, or if we will only support **0.8.6** and **Melodic**.

**0.8.6** is available in the default ros melodic packages. Versions **0.8.0** -> **0.8.5** were versions built for Melodic, but only available on **ros-shadow-fixed** and as we build a new system to release a binary version of our drivers to the public build farm.

https://github.com/fetchrobotics/fetch_robots/commit/52ac154a25a88a93de0151861c9db9451e6aeba2

moriarty, 2019-06-13:
Sorry-

We didn't get around to fixing this and releasing **0.7.29** into Indigo, which is now End of Life.

You should be able to upgrade to **18.04** and **Melodic** which will give you a new version of the drivers where this has been fixed.

https://docs.fetchrobotics.com/indigo_to_melodic.html

moriarty, 2019-08-20:
Re-opening this to check if it's actually fixed in *0.7.29* and later releases.

moriarty, 2019-08-20:
I'm closing this again because. I've tested on **Fetch3** with drivers 0.8.6

**amoriarty@fetch3:~$ rostopic echo /odom | grep frame_id -n1 -m3**
```
5-    nsecs: 179094103
6:  frame_id: "odom"
7:child_frame_id: "base_link"
8-pose: 
--
36-    nsecs: 188292249
37:  frame_id: "odom"
38-child_frame_id: "base_link"
```

**amoriarty@fetch3:~$ dpkg --status ros-melodic-fetch-drivers**
```
Package: ros-melodic-fetch-drivers
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 19814
Maintainer: Alexander Moriarty <amoriarty@fetchrobotics.com>
Architecture: amd64
Version: 0.8.6-0bionic.20190320.173739
Depends: libboost-filesystem1.65.1, libboost-program-options1.65.1, libboost-system1.65.1, libboost-thread1.65.1, libc6 (>= 2.14), libconsole-bridge0.4, libcurl4 (>= 7.16.2), libgcc1 (>= 1:3.0), libstdc++6 (>= 5.2), libtinyxml2-6 (>= 5.0.0), liburdfdom-world, libyaml-cpp0.5v5, curl, libboost-all-dev, libcurl4-openssl-dev, liburdfdom-dev, libyaml-cpp-dev, python-dev, ros-melodic-actionlib, ros-melodic-actionlib-msgs, ros-melodic-diagnostic-msgs, ros-melodic-fetch-auto-dock-msgs, ros-melodic-fetch-driver-msgs, ros-melodic-nav-msgs, ros-melodic-power-msgs, ros-melodic-robot-calibration-msgs, ros-melodic-robot-controllers, ros-melodic-robot-controllers-interface, ros-melodic-rosconsole, ros-melodic-roscpp, ros-melodic-roscpp-serialization, ros-melodic-rostime, ros-melodic-sensor-msgs, ros-melodic-urdf
Description: The public fetch_drivers package is a binary only release.
 fetch_drivers contains both the drivers and firmware for the fetch and freight research robots. There should be no reason to use these drivers unless you're running on a fetch or a freight research robot. This package, is a cmake/make only package which installs the binaries for the drivers and firmware.
Homepage: https://wiki.ros.org/fetch_drivers

```

moriarty, 2019-08-20:
I need more :coffee: .. I checked the wrong topic :man_facepalming:  sorry.

This issue is still open :-1:

moriarty, 2019-08-20:
**amoriarty@fetch3:~rostopic echo /imu | grep frame_id -n2 -m3**
```
4-    secs: 1566323843
5-    nsecs: 933911997
6:  frame_id: ''
7-orientation: 
8-  x: 0.0
--
27-    secs: 1566323843
28-    nsecs: 943921238
29:  frame_id: ''
30-orientation: 
31-  x: 0.0
--
50-    secs: 1566323843
51-    nsecs: 953972405
52:  frame_id: ''
53-orientation: 
54-  x: 0.0
```

moriarty, 2019-08-20:
Found the problem.

The frame_id is being set for `/imu1/imu` and `/imu2/imu` but not `/imu` which is the combined imu.

The constructor should have a better default for the frame.

closed 2019-08-21 by moriarty

moriarty, 2019-08-21:
This was automatically closed when I merged a commit to the private repo mentioning this issue.

The real fix is in the CI pipeline and will be available in **0.8.8**

moriarty, 2019-08-21:
binary drivers 0.8.8 https://github.com/ros/rosdistro/pull/22072 should fix this.