fetchrobotics/fetch_ros #71: Installation instructions?

DanielTakeshi, 2018-03-23:
I'm new to ROS and am wondering about the installation for these packages. Are these supposed to be built in any way, or can we just copy these in, for instance, where the `roscd` command brings me to? Thanks.

erelson, 2018-03-23:
These packages are made available as debians (so you can use apt-get to install them) via packages.fetchrobotics.com.  You can see the apt-get setup that references this on a robot by looking at /etc/apt/sources.list.d/[fetch|ros]-latest.list.  Note that at present we only support (and actively work with) Ubuntu 14.04.  Hope that helps.

DanielTakeshi, 2018-03-23:
Thanks @erelson for the helpful response. I am indeed using Ubuntu 14.04 and ROS Indigo.

I have a `ros-latest.list` but no `fetch-latest.list`:

```
(py2-fetch)daniel@daniel-ubuntu-mac:~$ ls -lh /etc/apt/sources.list.d/
total 4.0K
-rw-r--r-- 1 root root 51 Mar 22 16:48 ros-latest.list
(py2-fetch)daniel@daniel-ubuntu-mac:~$ cat /etc/apt/sources.list.d/ros-latest.list 
deb http://packages.ros.org/ros/ubuntu trusty main
```

So I made the fetch-latest file:

```
(py2-fetch)daniel@daniel-ubuntu-mac:~$ sudo vim /etc/apt/sources.list.d/fetch-latest.list 
(py2-fetch)daniel@daniel-ubuntu-mac:~$ cat /etc/apt/sources.list.d/fetch-latest.list 
deb http://packages.fetchrobotics.com/ros/ubuntu trusty main
```

Then `sudo apt-get update` seems to run without errors:

```
(py2-fetch)daniel@daniel-ubuntu-mac:~$ sudo apt-get update
Ign http://packages.fetchrobotics.com trusty InRelease
Get:1 http://packages.fetchrobotics.com trusty Release.gpg [181 B]             
Get:2 http://packages.fetchrobotics.com trusty Release [3,801 B]               
Get:3 http://packages.fetchrobotics.com trusty/main amd64 Packages [838 kB]    
Ign http://us.archive.ubuntu.com trusty InRelease                              
Hit http://us.archive.ubuntu.com trusty-updates InRelease                      
Hit http://packages.ros.org trusty InRelease                                   
Get:4 http://packages.fetchrobotics.com trusty/main i386 Packages [837 kB]     
Ign http://extras.ubuntu.com trusty InRelease                                  
Hit http://security.ubuntu.com trusty-security InRelease                       
Hit http://packages.ros.org trusty/main amd64 Packages                         
Hit http://us.archive.ubuntu.com trusty-backports InRelease                    
Hit http://packages.ros.org trusty/main i386 Packages                          
Hit http://us.archive.ubuntu.com trusty Release.gpg                            
Hit http://extras.ubuntu.com trusty Release.gpg                                
Hit http://security.ubuntu.com trusty-security/main Sources                    
Hit http://us.archive.ubuntu.com trusty-updates/main Sources                   
Ign http://packages.fetchrobotics.com trusty/main Translation-en_US            
Ign http://packages.fetchrobotics.com trusty/main Translation-en               
Hit http://us.archive.ubuntu.com trusty-updates/restricted Sources             
Hit http://extras.ubuntu.com trusty Release                                    
Hit http://security.ubuntu.com trusty-security/restricted Sources              
Hit http://us.archive.ubuntu.com trusty-updates/universe Sources               
Ign http://packages.ros.org trusty/main Translation-en_US                      
Hit http://extras.ubuntu.com trusty/main Sources                               
Hit http://us.archive.ubuntu.com trusty-updates/multiverse Sources             
Ign http://packages.ros.org trusty/main Translation-en                         
Hit http://security.ubuntu.com trusty-security/universe Sources                
Hit http://us.archive.ubuntu.com trusty-updates/main amd64 Packages            
Hit http://extras.ubuntu.com trusty/main amd64 Packages                        
Hit http://security.ubuntu.com trusty-security/multiverse Sources              
Hit http://us.archive.ubuntu.com trusty-updates/restricted amd64 Packages
Hit http://us.archive.ubuntu.com trusty-updates/universe amd64 Packages        
Hit http://extras.ubuntu.com trusty/main i386 Packages                         
Hit http://security.ubuntu.com trusty-security/main amd64 Packages             
Hit http://us.archive.ubuntu.com trusty-updates/multiverse amd64 Packages      
Hit http://us.archive.ubuntu.com trusty-updates/main i386 Packages             
Hit http://security.ubuntu.com trusty-security/restricted amd64 Packages       
Hit http://us.archive.ubuntu.com trusty-updates/restricted i386 Packages       
Hit http://us.archive.ubuntu.com trusty-updates/universe i386 Packages         
Hit http://security.ubuntu.com trusty-security/universe amd64 Packages         
Hit http://us.archive.ubuntu.com trusty-updates/multiverse i386 Packages       
Hit http://us.archive.ubuntu.com trusty-updates/main Translation-en            
Hit http://security.ubuntu.com trusty-security/multiverse amd64 Packages       
Hit http://us.archive.ubuntu.com trusty-updates/multiverse Translation-en      
Hit http://security.ubuntu.com trusty-security/main i386 Packages              
[... 45 lines cut ...]
Ign http://us.archive.ubuntu.com trusty/multiverse Translation-en_US
Ign http://us.archive.ubuntu.com trusty/restricted Translation-en_US
Ign http://us.archive.ubuntu.com trusty/universe Translation-en_US             
Fetched 1,679 kB in 6s (276 kB/s)                                              
Reading package lists... Done
```

Unfortunately apt-get cannot seem to find the packages:

```
(py2-fetch)daniel@daniel-ubuntu-mac:~$ sudo apt-get install fetch_ros
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package fetch_ros
(py2-fetch)daniel@daniel-ubuntu-mac:~$ sudo apt-get install fetch_navigation
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package fetch_navigation
```

(I assume we're supposed to use `fetch_navigation` if we want to install those specific things, not `fetch_ros` as well?)

Is there an obvious step I am missing?

erelson, 2018-03-23:
This should fix it for you!: http://docs.fetchrobotics.com/care_and_feeding.html#re-setting-up-apt-get-sources

Note, I don't know the exact provenance of our ros-latest mirror; I think we update it when we release new debians, which we last did at the end of December.  For best effects I suggest using our mirror, but I'm not sure whether you'll actually have issues if you use the main mirror (with presumably a few slightly newer packages).

DanielTakeshi, 2018-03-23:
Thanks, but from those two commands, this is what I end up getting on my laptop (note: this is not the robot itself, but my laptop which interfaces with it):

```
daniel@daniel-ubuntu-mac:~$ cat /etc/apt/sources.list.d/fetch-latest.list 
deb http://packages.fetchrobotics.com/fetch/ubuntu trusty main
daniel@daniel-ubuntu-mac:~$ cat /etc/apt/sources.list.d/ros-latest.list 
deb http://packages.fetchrobotics.com/ros/ubuntu trusty main
daniel@daniel-ubuntu-mac:~$
```

So the only difference between what I had before and what I have now is that the `ros-latest.list` file refers to a different URL. Earlier that list referred to `deb http://packages.ros.org/ros/ubuntu trusty main`.

Just to be absolutely clear:

(1) This is designed to be on the computers that interface with the robot, not the robot itself? We'll be running lots of stuff on our external machines with better processors, etc., and I was trying to follow the Fetch tutorials on my laptop which uses some of these libraries.

(2) The commands we should use are `sudo apt-get install fetch_navigation` or something like that? No extra fancy URLs or stuff appended?

mikeferguson, 2018-03-23:
You'll never be able to apt-get "fetch_navigation" because the debian is actually called ros-indigo-fetch-navigation or ros-indigo-fetch-ros (same rules apply to all ROS debs -- name is ros-Distro-X where X = package name, with "_" replaced with "-").

From the above, I think your sources are right, you just need to use the Debian package name rather than the ROS package name.

DanielTakeshi, 2018-04-02:
Thanks @mikeferguson  it seems like that did the trick. I actually just tried installing the gazebo and it installed the stuff here as a side effect.

```
daniel@daniel-ubuntu-mac:~$ sudo apt-get install ros-indigo-fetch-gazebo-demo
[sudo] password for daniel: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libamd2.3.1 libasound2-dev libavahi-client-dev libavahi-common-dev
  libblas-dev libbtf1.2.0 libcaca-dev libcamd2.3.1 libccolamd2.8.0
  libcholmod2.1.2 libcsparse3.1.2 libcxsparse3.1.2 libdbus-1-dev libklu1.2.1
  liblapack-dev libldl2.1.0 libpulse-dev libsdl-image1.2 libsdl-image1.2-dev
  libsdl1.2-dev libslang2-dev libspqr1.3.1 libsuitesparse-dev libumfpack5.6.2
  libwebp-dev libwebpdemux1 ros-indigo-amcl ros-indigo-base-local-planner
  ros-indigo-clear-costmap-recovery ros-indigo-control-toolbox
  ros-indigo-costmap-2d ros-indigo-fetch-depth-layer
  ros-indigo-fetch-description ros-indigo-fetch-gazebo
  ros-indigo-fetch-ikfast-plugin ros-indigo-fetch-maps
  ros-indigo-fetch-moveit-config ros-indigo-fetch-navigation
  ros-indigo-grasping-msgs ros-indigo-map-server ros-indigo-move-base
  ros-indigo-move-base-msgs ros-indigo-moveit-python ros-indigo-nav-core
  ros-indigo-navfn ros-indigo-open-karto ros-indigo-opencv-candidate
  ros-indigo-realtime-tools ros-indigo-rgbd-launch
  ros-indigo-robot-controllers ros-indigo-robot-controllers-interface
  ros-indigo-robot-controllers-msgs ros-indigo-rotate-recovery
  ros-indigo-simple-grasping ros-indigo-slam-karto
  ros-indigo-sparse-bundle-adjustment ros-indigo-voxel-grid
Suggested packages:
  libasound2-doc
The following NEW packages will be installed:
  libamd2.3.1 libasound2-dev libavahi-client-dev libavahi-common-dev
  libblas-dev libbtf1.2.0 libcaca-dev libcamd2.3.1 libccolamd2.8.0
  libcholmod2.1.2 libcsparse3.1.2 libcxsparse3.1.2 libdbus-1-dev libklu1.2.1
  liblapack-dev libldl2.1.0 libpulse-dev libsdl-image1.2 libsdl-image1.2-dev
  libsdl1.2-dev libslang2-dev libspqr1.3.1 libsuitesparse-dev libumfpack5.6.2
  libwebp-dev libwebpdemux1 ros-indigo-amcl ros-indigo-base-local-planner
  ros-indigo-clear-costmap-recovery ros-indigo-control-toolbox
  ros-indigo-costmap-2d ros-indigo-fetch-depth-layer
  ros-indigo-fetch-description ros-indigo-fetch-gazebo
  ros-indigo-fetch-gazebo-demo ros-indigo-fetch-ikfast-plugin
  ros-indigo-fetch-maps ros-indigo-fetch-moveit-config
  ros-indigo-fetch-navigation ros-indigo-grasping-msgs ros-indigo-map-server
  ros-indigo-move-base ros-indigo-move-base-msgs ros-indigo-moveit-python
  ros-indigo-nav-core ros-indigo-navfn ros-indigo-open-karto
  ros-indigo-opencv-candidate ros-indigo-realtime-tools ros-indigo-rgbd-launch
  ros-indigo-robot-controllers ros-indigo-robot-controllers-interface
  ros-indigo-robot-controllers-msgs ros-indigo-rotate-recovery
  ros-indigo-simple-grasping ros-indigo-slam-karto
  ros-indigo-sparse-bundle-adjustment ros-indigo-voxel-grid
0 upgraded, 58 newly installed, 0 to remove and 14 not upgraded.
Need to get 16.2 MB of archives.
After this operation, 74.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] y

(and then it installed...)
```

The calibration package was the only one that wasn't additionally installed, but it was straightforward to use the installation once I realized you use dashes and the `ros-indigo-[...]` when doing apt get install. 

I think this is resolved.

closed 2018-04-02 by DanielTakeshi