fetchrobotics/fetch_ros #72: "Can't call nice to prioritize freight_driver!!!"

BillWSY, 2018-04-20:
Hi Fetch,

I have been playing with fetch_driver in a Linux containers, and I run into the `Can't call nice to prioritize freight_driver!!!`.

I have tried adding a few methods, and I can even set priority with `renice`, but the error is still there. ** Since I don't have access to the source code, can help me identify what is the logic behind this -- when will fetch_driver print this error message? **

Thank you!

dbking77, 2018-04-21:
A recent change was made to fetch_drivers so they ask for a higher "nice"
priority when first starting.   If the drivers cannot get this higher
priority it complains, but will run the same as before, so I think it is
safe to ignore it.

If you look at the standard robot upstart, it gives the robot driver access
to a better nice level with the following lines:
limit memlock unlimited unlimited
limit nice 35 35

-Derek

BillWSY, 2018-04-21:
Thank you. I was able to set ulimit levels for my containers and the driver is not complaining anymore.

closed 2018-04-21 by BillWSY