fetchrobotics/fetch_ros #74: How to obtain three out of the five head camera image topics "of interest" (from the docs)?

DanielTakeshi, 2018-05-03:
In the Fetch docs:

http://docs.fetchrobotics.com/api_overview.html#head-camera-interface

on the head camera interface, it is stated that there are five topics of interests.

When I do the following, though, it looks like only two out of the five happen to appear. (I'm using the standard ROS Indigo and Ubuntu 14.04.)

First, I start by launching a simulator:

```
roslaunch fetch_gazebo playground.launch
```

and looks like the simulator is working. I move the Fetch around to get more interesting camera views:

[image]

I next run `python camera_simple.py` where the `camera_simple.py` script is as follows:

```
import argparse, cv2, os, rospy, sys, threading
from sensor_msgs.msg import Image, JointState, PointCloud2
from cv_bridge import CvBridge, CvBridgeError

class SimpleData(object):

    def __init__(self):
        self._msg = None

    def __call__(self, data):
        self._msg = data

    def get_data(self):
        return self._msg

def main(args):
    bridge = CvBridge()
    rospy.init_node('main', anonymous=True)

    topics = [ 
        ('head_camera/depth_registered/points', PointCloud2),
        ('head_camera/depth_downsampled/points', PointCloud2), # None :(
        ('head_camera/depth/image_raw', Image), # None :(
        ('head_camera/depth/image', Image), # None :(
        ('head_camera/rgb/image_raw', Image),
    ]
    data_collectors = [SimpleData() for _ in range(len(topics))]

    for idx,item in enumerate(topics):
        rospy.Subscriber(name=item[0],
                         data_class=item[1],
                         callback=data_collectors[idx], 
                         queue_size=1)
    rospy.sleep(2)

    while not rospy.is_shutdown():
        data = [data_collectors[i].get_data() for i in range(len(data_collectors))]
        depth = data[0]
        imgraw = data[4]
        for item in data:
            print(type(item))
        img = bridge.imgmsg_to_cv2(imgraw, "bgr8")
        name = 'test.png'
        cv2.imshow(name, img)
        cv2.imwrite(name, img)
        cv2.waitKey(3)

if __name__ == '__main__':
    parser = argparse.ArgumentParser()
    parser.add_argument('--robot', type=str, default='fetch')
    parser.add_argument('--imdir', type=str, default='images/')
    args = parser.parse_args()
    main(args)
```

However, the result I get is:
[image]

The good news is that I can see the camera image topic `head_camera/rgb/image_raw` and the point cloud one `head_camera/depth_registered/points`. However it seems like the other three are `None`.

How do I obtain data for those other three topics? 

(On a side note, is there a way to convert the PointCould2 to a simple png image that I can then save for later?[ I haven't found anything from the ROS docs...](http://docs.ros.org/api/sensor_msgs/html/msg/PointCloud2.html)) 

Thanks for your help!

DanielTakeshi, 2018-05-04:
Figured this out somewhat, the docs seem to be outdated. After running the gazebo, I ran `rostopic list` and got this for all the `head_camera` stuff:

```
/head_camera/crop_decimate/parameter_descriptions
/head_camera/crop_decimate/parameter_updates
/head_camera/depth_downsample/camera_info
/head_camera/depth_downsample/image_raw
/head_camera/depth_downsample/image_raw/compressed
/head_camera/depth_downsample/image_raw/compressed/parameter_descriptions
/head_camera/depth_downsample/image_raw/compressed/parameter_updates
/head_camera/depth_downsample/image_raw/compressedDepth
/head_camera/depth_downsample/image_raw/compressedDepth/parameter_descriptions
/head_camera/depth_downsample/image_raw/compressedDepth/parameter_updates
/head_camera/depth_downsample/image_raw/theora
/head_camera/depth_downsample/image_raw/theora/parameter_descriptions
/head_camera/depth_downsample/image_raw/theora/parameter_updates
/head_camera/depth_downsample/points
/head_camera/depth_registered/camera_info
/head_camera/depth_registered/image_raw
/head_camera/depth_registered/points
/head_camera/head_camera_nodelet_manager/bond
/head_camera/parameter_descriptions
/head_camera/parameter_updates
/head_camera/rgb/camera_info
/head_camera/rgb/image_raw
/head_camera/rgb/image_raw/compressed
/head_camera/rgb/image_raw/compressed/parameter_descriptions
/head_camera/rgb/image_raw/compressed/parameter_updates
/head_camera/rgb/image_raw/compressedDepth
/head_camera/rgb/image_raw/compressedDepth/parameter_descriptions
/head_camera/rgb/image_raw/compressedDepth/parameter_updates
/head_camera/rgb/image_raw/theora
/head_camera/rgb/image_raw/theora/parameter_descriptions
/head_camera/rgb/image_raw/theora/parameter_updates
```

I switched the names around a bit and got some images which I could save.

I suppose that resolves that issue. I still wonder if there's a way to save PointCloud2 data to png?

mikeferguson, 2018-05-04:
It appears that API documentation is really only accurate for the real robot -- the gazebo interface is close (there are no IMU or breakers either AFAIK). If I recall correctly, the Gazebo model uses the default 3d camera model, which doesn't support depth images -- which would knock out the two topics you don't see.

A PR to add the information about what the proper simulation data is would probably be appreciated.

For saving your images, I think you'll manually have to translate the pixels from PointCloud2 to Image datatypes. I'm sure there is something cleaner than this (and free of PCL), but here's a pointer to such a thing: https://github.com/mikeferguson/chessbox/blob/indigo-devel/chess_perception/include/chess_perception/conversions.h

DanielTakeshi, 2018-05-08:
Thanks @mikeferguson , so the difference in gazebo vs real robot means my collaborators and I will most likely add an option to our code to specify whether we're using the simulator or a real robot. I'll report back on any other differences that I encounter, though hopefully there won't be any.

Sorry, just another quick question, is there additional documentation on what the topics mean, besides the ones that are listed in the fetch api here: http://docs.fetchrobotics.com/api_overview.html ?

DanielTakeshi, 2018-05-16:
I tested on the real robot, and the topics seem to match the Gazebo simulator exactly, as in the docs are a bit out of date. However it's clear that one should use the names as resulted from `rostopic list` anyway.

closed 2018-05-16 by DanielTakeshi