fetchrobotics/fetch_ros #75: support for robot_calibration updates

mikeferguson, 2018-06-05:
As an FYI, I pretty much rewrote how finders and optimization work in robot_calibration, see:
 * [Finders are now plugin-based](https://github.com/mikeferguson/robot_calibration/issues/37)
 * [GroundPlane/PlaneFinder are now merged into single finder with unified params](https://github.com/mikeferguson/robot_calibration/issues/38)
 * Error block names/params have been cleaned up -- see [Readme](https://github.com/mikeferguson/robot_calibration#updating-from-indigo)

I have an (untested) branch for updating the Fetch to this new release which is in Kinetic+Melodic: https://github.com/mikeferguson/fetch_ros/tree/melodic-devel

moriarty, 2019-06-13:
Sorry I added a milestone but still forgot

mikeferguson, 2019-11-20:
I actually tested this today

closed 2019-11-20 by mikeferguson