fetchrobotics/fetch_ros #84: ps3joy / sixad issues

moriarty, 2018-07-27:
See:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1764645

There is an upstream patch coming... It's upstream in master and bionic.
https://lists.ubuntu.com/archives/kernel-team/2018-June/093557.html
https://lists.ubuntu.com/archives/kernel-team/2018-June/093434.html

To work around this in the meantime, I used the patch from here (link was in bug report)
https://github.com/jeremyb31/newbtfix-4.15

We don't really want to require maintaining any old sixad stuff, and want to use the out-of-the box kernel drivers.

- [ ] The patch officially available- I hope this doesn't take _too_ long, I don't want this to delay the release.
- [ ] Need to find better params for the ps3joy so that you don't need to keep the joypad moving all the time.
- [ ] Need new documentation for the joypad... this PR https://github.com/fetchrobotics/fetch_ros/pull/61 changed the buttons

moriarty, 2018-08-19:
It looks like the patch might finally by in the ubuntu bionic kernel 4.15.0-33.

erelson, 2019-08-12:
We weren't able to find a fix to make the PS3 controller fully usable. We've moved instead to recommending PS4 controllers for 18.04 robots.

closed 2019-08-12 by erelson