fetchrobotics/fetch_ros #93: Please consider using a more permissive license than CC-BY-NC-ND for `fetch_description`

alimuldal, 2018-09-05:
As far as open source licenses go, CC-BY-NC-ND is about as restrictive as it gets. In particular, the "no derivative works" clause means that it's not possible for others to distribute versions of the URDFs/meshes that have been modified for research purposes, or converted to different formats (e.g. MuJoCo MJCF). Would you consider switching to a more permissive open source license, such as BSD or Apache?

erelson, 2018-09-06:
We'll look into whether this is the appropriate license for this. Thanks for the heads-up @alimuldal !

@moriarty: https://github.com/fetchrobotics/fetch_ros/blob/melodic-devel/fetch_description/package.xml#L13
https://github.com/fetchrobotics/fetch_ros/commit/505c51bd61b16af2ecd758a17afe5d35bc318bff

alimuldal, 2018-10-23:
Hi @moriarty, any news on this?

alimuldal, 2018-10-26:
I'm one of the developers of the DeepMind Control Suite (https://github.com/deepmind/dm_control). I'm interested in potentially using the Fetch meshes and URDFs to create some reinforcement learning environments for research purposes. I'd need to convert the URDFs to MuJoCo's MJCF format, and probably make some tweaks to the model and/or meshes in order to get the desired behaviour in MuJoCo. The "ND" clause means that I would not be able to redistribute the MJCFs or modified meshes. The "NC" clause is also problematic, since I work for Google and everything we do may be considered "commercial" (including research).

moriarty, 2018-10-30:
Okay thank you for the clarification.

I have spoken with @mmwise and removing the "ND" clause is something that we are considering.

> The "NC" clause is also problematic, since I work for Google and everything we do may be considered "commercial" (including research).

I don't think this is something that we'll remove.

From your description, it sounds like removing the "ND" clause is what is blocking you from going forward and you would need the "NC" clause removed only as a legal precaution in the event that Google decides to commercialize this research?

Would removing the "ND" clause be a reasonable compromise?

alimuldal, 2018-10-30:
Removing the "no derivatives" clause seems like it would be a helpful change for academic researchers who might want to build on these models, but unfortunately our company policy means that I can't touch any code released under a non-commercial license (for any purpose, including research that's unrelated to any product).

moriarty, 2018-11-15:
Just to ping @alimuldal. We're still talking about this internally.

mmwise, 2019-07-13:
Selected License
Attribution-NonCommercial-ShareAlike 4.0 International

moriarty, 2019-07-25:
@alimuldal I apologize that this took so long, and hope you're able to share some of your work, despite the **NC** clause still being there. 

> I'm one of the developers of the DeepMind Control Suite (https://github.com/deepmind/dm_control). I'm interested in potentially using the Fetch meshes and URDFs to create some reinforcement learning [...]

I'm sure the MuJoCo's would be appreciated by the community

closed 2019-07-25 by moriarty