fetchrobotics/fetch_ros #102: Fetch navigation performs poorly in Melodic simulation

nickswalker, 2019-01-15:
**Steps**
With up to date versions of fetch_ros and fetch_gazebo

    roslaunch fetch_gazebo playground.launch

And

    roslaunch fetch_gazeo_demo fetch_nav.launch 

**Behavior**
When given a nav goal, the robot's localization drifts quickly (seems like it happens during rotation). The robot is never able to reach the goal.

https://youtu.be/w1y0b5aI3o8

Nothing jumps out from the standard `move_base` configurations so I'm not sure what's going on.

moriarty, 2019-01-27:
Thanks, we'll take a look.

FYI:
@velveteenrobot, @cjds @erelson & @narora1

moriarty, 2019-02-13:
@nickswalker sorry for the delay everyone I originally tags has been busy.

I've just spoken with @safrimus and he'll investigate.
I've also created an internal JIRA ticket in hopes of not loosing track of this issue again.
https://fetchrobotics.atlassian.net/browse/OPEN-31

cjds, 2019-02-13:
@nickswalker to clarify this only happens with melodic? and Gazebo 9?

nickswalker, 2019-02-13:
Yes, I have only observed this happening in Melodic with Gazebo 9.

dbking77, 2019-02-14:
To me, it looks like its related to localization.   Usually when you first
localize robot, its particle cloud is pretty spread out the estimated robot
position jumps around
at bit while the robot figures out where it is.
However, usually the particle cloud converges to the correct position, and
this jumpiness stops happening.   Take a look at the AMCL particle cloud
output in RViz (PoseArray type, can't remember what topic is)

-Derek

nickswalker, 2019-02-14:
Here are some clips with AMCL and the localization transforms visualized:

https://www.youtube.com/watch?v=uNb0pJbObHA

https://www.youtube.com/watch?v=sk4ANbCywUk

I pulled in all the recent Fetch changes and bumped to the latest Melodic sync.

It doesn't seem like the AMCL config, the Fetch Gazebo model, or any other component that might obviously cause localization to drift so quickly was changed between the indigo and melodic releases. But it's eminently reproduceable for me. I have a couple machines now where I can start a fresh workspace, clone everything, run the launch files and observe this behavior.

Let me know if bags would help.

moriarty, 2019-02-14:
@nickswalker thanks, @safrimus was also able to reproduce immediately in the simulator following your steps in the original issue.

I haven't seen this on the actual hardware, can you confirm that navigation is working on your fetch running melodic?

nickswalker, 2019-02-14:
Yes, navigation has been working fine on the real robot