fetchrobotics/fetch_ros #109: Could not find the opencv_candidateConfig.cmake

huangjiancong1, 2019-03-24:


huangjiancong1, 2019-03-24:
Ros-Kinetic with gazebo7:
package version: https://bitbucket.org/theconstructcore/fetch_tc/src/r0-kinetic/fetch_sim/ with the branch: ro-kinetic

Error:
```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'gazebo' but neither 'gazebo_INCLUDE_DIRS' nor
  'gazebo_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  fetch_tc/fetch_sim/fetch_gazebo/CMakeLists.txt:25 (catkin_package)

-- +++ processing catkin package: 'fetch_depth_layer'
-- ==> add_subdirectory(fetch_tc/fetch_sim/fetch_ros/fetch_depth_layer)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "opencv_candidate"
  with any of the following names:

    opencv_candidateConfig.cmake
    opencv_candidate-config.cmake

  Add the installation prefix of "opencv_candidate" to CMAKE_PREFIX_PATH or
  set "opencv_candidate_DIR" to a directory containing one of the above
  files.  If "opencv_candidate" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  fetch_tc/fetch_sim/fetch_ros/fetch_depth_layer/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'opencv_candidate'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "opencv_candidate"
  with any of the following names:

    opencv_candidateConfig.cmake
    opencv_candidate-config.cmake

  Add the installation prefix of "opencv_candidate" to CMAKE_PREFIX_PATH or
  set "opencv_candidate_DIR" to a directory containing one of the above
  files.  If "opencv_candidate" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  fetch_tc/fetch_sim/fetch_ros/fetch_depth_layer/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/jim/fetch_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jim/fetch_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

huangjiancong1, 2019-03-24:
When is use their other branch names **kinetic**, it can find the opencv_candidate but have miss some API:
```
[ 25%] Building CXX object fetch_tc/fetch_sim/opencv_candidate/src/rgbd/CMakeFiles/opencv_rgbd2.dir/src/odometry.cpp.o
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/opencv_candidate/datamatrix.cpp:4:37: fatal error: opencv2/legacy/compat.hpp: No such file or directory
compilation terminated.
In file included from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:36:0:
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:134:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:202:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:315:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
In file included from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:36:0:
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:502:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:551:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:605:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
fetch_tc/fetch_sim/opencv_candidate/src/opencv_candidate/CMakeFiles/opencv_candidate.dir/build.make:62: recipe for target 'fetch_tc/fetch_sim/opencv_candidate/src/opencv_candidate/CMakeFiles/opencv_candidate.dir/datamatrix.cpp.o' failed
make[2]: *** [fetch_tc/fetch_sim/opencv_candidate/src/opencv_candidate/CMakeFiles/opencv_candidate.dir/datamatrix.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:39:0:
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.h: In instantiation of ‘size_t convertDepthToFloat(const cv::Mat&, const cv::Mat&, float, cv::Mat_<float>&, cv::Mat_<float>&, cv::Mat_<float>&) [with T = short unsigned int; size_t = long unsigned int]’:
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:107:95:   required from here
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.h:80:20: error: call of overloaded ‘cvIsNaN(short unsigned int&)’ is ambiguous
         if (cvIsNaN(depth_i) || (depth_i == std::numeric_limits<T>::min()) || (depth_i == std::numeric_limits<T>::max()))
                    ^
In file included from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/saturate.hpp:49:0,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/matx.hpp:54,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core.hpp:57,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/core.hpp:48,
                 from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:43,
                 from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:36:
/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/fast_math.hpp:164:15: note: candidate: int cvIsNaN(double)
 CV_INLINE int cvIsNaN( double value )
               ^
/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/fast_math.hpp:252:15: note: candidate: int cvIsNaN(float)
 CV_INLINE int cvIsNaN( float value )
               ^
In file included from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:39:0:
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.h: In instantiation of ‘size_t convertDepthToFloat(const cv::Mat&, const cv::Mat&, float, cv::Mat_<float>&, cv::Mat_<float>&, cv::Mat_<float>&) [with T = short int; size_t = long unsigned int]’:
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:109:94:   required from here
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.h:80:20: error: call of overloaded ‘cvIsNaN(short int&)’ is ambiguous
         if (cvIsNaN(depth_i) || (depth_i == std::numeric_limits<T>::min()) || (depth_i == std::numeric_limits<T>::max()))
                    ^
In file included from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/saturate.hpp:49:0,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/matx.hpp:54,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core.hpp:57,
                 from /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/core.hpp:48,
                 from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:43,
                 from /home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/src/rgbd/src/depth_to_3d.cpp:36:
/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/core/fast_math.hpp:164:15: note: candidate: int cvIsNaN(double)
 CV_INLINE int cvIsNaN( double value )
               ^
[... cut ...]
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:502:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:551:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
/home/jim/fetch_ws/src/fetch_tc/fetch_sim/opencv_candidate/include/opencv2/rgbd/rgbd.hpp:605:5: error: ‘AlgorithmInfo’ does not name a type
     AlgorithmInfo*
     ^
Scanning dependencies of target cv_bridge

```

huangjiancong1, 2019-03-24:
Issue done by:
sudo apt-get install ros-kinetic-opencv-candidate and ros-kinetic-opencv-apps

huangjiancong1, 2019-03-24:
After finish compiled, i can not tun `roscore`, is seems that i without some `libcpp file`, like:
```
process[rosout-1]: started with pid [1476]
started core service [/rosout]
[rosout-1] process has died [pid 1476, exit code 127, cmd /opt/ros/kinetic/lib/rosout/rosout __name:=rosout __log:=/home/jim/.ros/log/7300ea94-4e01-11e9-9ee8-9061ae4896b2/rosout-1.log].
log file: /home/jim/.ros/log/7300ea94-4e01-11e9-9ee8-9061ae4896b2/rosout-1*.log
[rosout-1] restarting process

```

When I run: `ldd /opt/ros/kinetic/lib/rosout/rosout`:
```
libroscpp.so => not found
librosconsole.so => not found
libroscpp_serialization.so => not found
librostime.so => not found

```

moriarty, 2019-03-31:
1. That repository and branch is from TheConstructSim so I will tag @RDaneelOlivav It would have been best to raise the issue with where you found the source code.
2. It looks like you're trying to use the fetch in simulation (we only support 14.04 and 18.04) so this issue would be better in https://github.com/fetchrobotics/fetch_gazebo
3. It seems you've cloned a package but haven't installed the dependencies for the package
    `cd /home/jim/fetch_ws/`
    `rosdep install -y --from-paths src --ignore-src --rosdistro kinetic`
4. I recommend using the packages and repositories from @fetchrobotics which we release to the public ros build farm via @fetchrobotics-gbp - you can simply `apt-get install ros-melodic-fetch-simulation`
5. The reason `ldd /opt/ros/kinetic/lib/...` fails with the `libros... => not found` is because you haven't sourced the `/opt/ros/kinetic/setup.bash`

As the issues are not coming from code in our @fetchrobotics/opensource projects, I'll close this issue, it looks like the problem is in your ros setup.

the official "fetch_ros" package is passing on the buildfarm:
1. Melodic 18.04 http://build.ros.org/job/Mdev__fetch_ros__ubuntu_bionic_amd64/
2. Kinetic 16.04 http://build.ros.org/job/Kdev__fetch_ros__ubuntu_xenial_amd64/
3. Indigo 14.04 http://build.ros.org/job/Idev__fetch_ros__ubuntu_trusty_amd64/

closed 2019-03-31 by moriarty

RDaneelOlivav, 2019-04-01:
@huangjiancong1 As @moriarty said, it would be better to place the issues in the git you find them. Having said that, now the fetch_gazebo gits are being updated a lot, and have great improvements and starting to have support for Melodic, so I highly recommend that you use the fetch official git, because it has the latest changes. Theconstruct version is a stable version that we update only when the systems became really stable and more oriented to use for the RobotIgniteCourses. The one you choose depends on what you are looking for.