fetchrobotics/fetch_ros #112: Error compiling fetch_teleop

Zephyr06, 2019-03-31:
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>::subsecond_duration(const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&)
   class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                              ^~~~~~~~~~~~~~~~~~
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘float’ to ‘const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&’
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>::subsecond_duration(boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&&)
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘float’ to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&&’
CMakeFiles/joystick_teleop.dir/build.make:62: recipe for target 'CMakeFiles/joystick_teleop.dir/src/joystick_teleop.cpp.o' failed
make[2]: *** [CMakeFiles/joystick_teleop.dir/src/joystick_teleop.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/joystick_teleop.dir/all' failed
make[1]: *** [CMakeFiles/joystick_teleop.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'fetch_teleop': 
  Command '['/home/zephyr/catkin_ws/devel_isolated/fetch_simulation/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2

Any suggestions? Thanks a lot!

erelson, 2019-03-31:
Hi @Zephyr06, can you give more information on what OS/ROS release you're working with?

Zephyr06, 2019-03-31:
Hi, it is ROS melodic; It is a little strange because I tried to install and compile using the same procedure in another computer and it just worked.

erelson, 2019-04-01:
I would suggest comparing the boost and C++ compiler versions in use between the two computers to see what might be different. Hope that helps!

moriarty, 2019-04-04:
@Zephyr06 could you please provide details on the rest of your catkin workspace?

`wstool info` inside of your workspace/src directory and if you haven't set up `wstool` you must first run `wstool scrape` and add the repositories before running `wstool info`.

Also, please ensure all of your ros packages are up to date:
1. `sudo apt update`
2. `apt list --upgradable` you shouldn't have any ros packages which need to be updated.

Also, you're installing from source, so you'll first need to install the dependencies, in the root of your catkin workspace run:
`rosdep install -y --from-paths src --ignore-src --rosdistro melodic`

Zephyr06, 2019-04-08:
Well, I reinstall the total ubuntu computer system, and it works right now......

Zephyr06, 2019-04-08:
Thanks for all of you guys' help!

erelson, 2019-04-08:
Glad it's working!

closed 2019-04-08 by erelson