fetchrobotics/fetch_ros #117: [BUG] fetch.urdf links: bellows_link, bellows_link2 make robot bounce

josiahls, 2019-04-06:
**Describe the bug**
I am porting the fetch URDF to bullet3. Looking through the URDF I found bellows_link and bellows_link2. They are causing the robot to every-so-often jump due to self collision.

Main question, is it necessary to include a collision mesh for bellows_link2? It seems like it is just for visuals. 

When I comment out the bellows_link2 collision mesh, this problem is fixed. 

**To Reproduce**
```
Clone https://github.com/josiahls/pybullet-gym/tree/bug_bellows_link_hopping

cd pybulletgym/examples/fetch
python test_fetch_envs.py
```
After a few iterations the fetch robot hops.

**Expected behavior**
The robot stays completely still...

**Screenshots**
[image]

**catkin workspace (please complete the following information):**
Using pybullet3

**Additional context**
As I said, commenting out the collision meshe for the bellows_link2 fixes the issue.

erelson, 2019-04-08:
Quick take: looking at the gif, the torso is going down lower than is physically possible.  Not sure if that's related, or just a side effect of the internal collisions you note.

Do you know whether the bellows collision link is moving with the torso? (I believe the visual portion does.)  Is it colliding with the base, or with other parts of the torso (skins, etc.)?

A few observations; I am not truly familiar with the URDF model, and especially the collision model, but I know the following:
- we of course don't want the robot to hit the bellows in real life
- ideally, the bellows would expand/compress like on the real robot, but that wasn't feasible (and probably still isn't) in simulation.  So instead we have this approximation that initially worked for rviz and gazebo(?).

moriarty, 2019-07-25:
FYI We've now removed the **ND** clause from the license to allow for derivative works as long as they're still **NC** and **SA**.

I haven't had a chance to look into this issue yet.