fetchrobotics/fetch_ros #127: How disable moveit collision checking on fetch's gripper?

xiaoyuchhusky, 2019-08-13:
Hi all,

I would like to use moveit group to do motion planning for fetch's arm without considering the collision on its gripper.

I first enable the moveit active sensing, and then create the move_group by either:

`move_group = MoveGroupInterface("arm", "base_link")`
or
`move_group = moveit_commander.MoveGroupCommander("arm")`

However, the motion planner will always consider the collision between the gripper and the environment. 

1) Is there a way to disable the collision checking on gripper?
2) Where can I find the definition of the link group "arm" and "arm_with_torso"?

Thank you very much!

Best,
Yuchen

moriarty, 2019-08-14:
I guess I'll need to make a github issue template for questions... Please let me know the basics:

1. fetch_ros version:
2. OS version:
3. ROS version:
4. current catkin workspace information `wstool info`:
5. (Hardware) Robot Number
5. (Simulation) Gazebo Version / fetch_gazebo version

xiaoyuchhusky, 2019-08-14:
@moriarty Hi Alex,

I am using

1. fetch_ros version: 0.7.13
2. OS version: Ubuntu 14.04
3. ROS version: indigo
4. My catkin workspace named with "projects", I run
`wstool status -t projects/`
I got
`ERROR in config: /home/yuchen/projects has no workspace configuration file '.rosinstall'`
I guess this is because my workspace is not built by using wstool, right?
5. Robot Number: 1501-0000(MODEL)
6. fetch_gazebo version: 0.7.1

I am working on robot grasping. What I want to achieve is that, under moveit active sensing,

1) enable collision checking on both arm and gripper before the robot grasps the object;
2) disable the collision checking on the gripper after grasping an object.

Is this possible in python script?

Besides, I was also wondering how to give a namespace to the real fetch robot. The robot automatically publishes topics when it boots, so it doesn't work that just setting namespace variable in '.bashrc' file.

Thank you!

moriarty, 2019-09-11:
Okay, sorry for the delayed response.

1. You'll still want to consider collisions with the gripper. Otherwise you could damage the robot.

What you'll want is to allow collisions between the gripper and the object which has been grasped, and move the object which has been grasped from a collision object to a attached object.

the 'touch_links' argument is what you'll want to set this will allow some links to touch but not to completely disable collision checking.

http://docs.ros.org/melodic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a0ed3e5206cb80fea88f17668cfccb7c8

https://github.com/ros-planning/moveit/blob/8fcff1db6d4eb7c04236ad532f0e54b2804a6b6f/moveit_commander/src/moveit_commander/move_group.py#L554

There is also a moveit_python package which has wrapper functions for this. I recommend searching for `touch_links` as you'll need to set those.

https://github.com/mikeferguson/moveit_python/blob/master/src/moveit_python/planning_scene_interface.py

We recently had an intern make a demo, but her internship time finished before the code was completely documented
https://opensource.fetchrobotics.com/icra-challenge/2019/01/28/tutorial.html

https://github.com/fetchrobotics/fetch_demos/blob/0b5a95512846e65a737b065a1a742f2dac294997/fetch_demos_common/src/fetch_demos_common/fetch_grasping.py#L115