fetchrobotics/fetch_ros #138: [BUG] Fetch in Gazebo gets dark.

708yamaguchi, 2020-04-14:
## Bug overview
When I use fetch in gazebo, fetch in gazebo looks darker than expected. Fetch in Rviz looks normal.
I think the cause is this commit (https://github.com/fetchrobotics/fetch_ros/commit/20cd76f4ee2ee1796b36bae3191a1a8b4b959ef7 , next commit to `0.8.0`)
Could you please replace current `.dae` files with `.dae` files which does not look dark in Gazebo?

## Reproduce the situation
I use `Ubuntu 18.04` and `ROS melodic`.

#### Dark fetch:
```
mkdir ~/catkin_ws/src -p
cd ~/catkin_ws/src
git clone https://github.com/fetchrobotics/fetch_ros.git
cd fetch_ros
git checkout 20cd76f4ee2ee1796b36bae3191a1a8b4b959ef7 # Next commit to 0.8.0
cd ..
rosdep install --from-paths . --ignore-src -y -r
cd ..
source /opt/ros/melodic/setup.bash
catkin build
source ~/catkin_ws/devel/setup.bash
roslaunch fetch_gazebo playground.launch
```
|Gazebo|Rviz|
|---|---|
|[image]|[image]|

#### Normal fetch:
```
mkdir ~/catkin_ws/src -p
cd ~/catkin_ws/src
git clone https://github.com/fetchrobotics/fetch_ros.git
cd fetch_ros
git checkout 0.8.0
cd ..
rosdep install --from-paths . --ignore-src -y -r
cd ..
source /opt/ros/melodic/setup.bash
catkin build
source ~/catkin_ws/devel/setup.bash
roslaunch fetch_gazebo playground.launch
```
|Gazebo|Rviz|
|---|---|
|[image]|[image]|

erelson, 2020-04-16:
Hi @708yamaguchi, thanks for the report!

I noticed this as well last month, but only during a quick test, and assumed it was an issue with my laptop's GPU (GTX 1060, I believe).  Do you know if your setup uses an AMD or NVidia GPU? I'll check if others can confirm.  If it's an issue with the .dae files my initial assumption would be the material properties need to be updated in some manner.

708yamaguchi, 2020-04-16:
Hi @ereson, thank you very much for your support!

My laptop has NVIDIA GPU (details are below), but it seems that my laptop does not use GPU.
I checked this situation by the commands below. During playground.launch, GPU did not use its memory.
```bash
# Terminal 1
watch -n1 nvidia-smi
# Terminal 2
roslaunch fetch_gazebo playground.launch
```

Here is my CPU and GPU environment.
CPU
```bash
$ sudo lshw -class processor
  *-cpu                     
       description: CPU
       product: Intel(R) Core(TM) i7-6600U CPU @ 2.60GHz
       vendor: Intel Corp.
       physical id: 7
       bus info: cpu@0
       version: Intel(R) Core(TM) i7-6600U CPU @ 2.60GHz
       serial: None
       slot: U3E1
       size: 2037MHz
       capacity: 4005MHz
       width: 64 bits
       clock: 100MHz
       capabilities: x86-64 fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d cpufreq
       configuration: cores=2 enabledcores=2 threads=4
```
GPU
```bash
$ lspci | grep -i nvidia
06:00.0 3D controller: NVIDIA Corporation GM108M [GeForce 930M] (rev a2)
```

mikeferguson, 2020-04-16:
So we've definitely run into issues like this before. Basically, Gazebo and RVIZ load the shading information entirely differently -- we struck a fine balance with the original model to make it work the same in Gazebo and RVIZ.

708yamaguchi, 2020-04-16:
@mikeferguson 
Thank you very much for your information.

I understood the circumstances.

erelson, 2020-04-16:
@RDaneelOlivav do you have any insights to provide on this issue's topic? It appears that the changes in your PR last year for OpenWebGL/TheConstruct, had undesired effects in gazebo appearances.  Have you dealt with this for other robots you've simulated, or have recommendations?  Thanks!

RDaneelOlivav, 2020-04-16:
@erelson Hi. Well that could be for a number of reasons. In my experience when a texture apears black could be because:
1) When exporting the dae, it wasn't done correctly with lighting included in the blender scene and that could affect. It also helps to use models with the shadows disconnected, that way the textures are rendered similar to RVIZ.
2) This could also be due to not finding the textures when launching gazebo. Gazebo and rviz don't use the same import system and this could be one reason.

erelson, 2020-06-01:
As a datapoint, I do reproduce the dark model as shown in the original post.  I then switched my laptop from integrated Intel graphics to using the NVidia card/drivers, and features do resolve, somewhat.  Still not how it was prior to the .dae models getting changed.  Pending an external conversation, I will probably take the simple approach of reverting the .dae models.
[image]

erelson, 2021-01-05:
I've put the proposed fixed from June into PRs #153 and #154 for noetic and melodic respectively.

708yamaguchi, 2021-01-05:
@erelson 
Thank you so much!

I can see bright fetch in Gazebo with your new `erelson:fix_urdf_textures_melodic` branch.

PR #154 (2021-01-08): Manually revert textures on URDF models
Fixes #138. Copied files from the melodic-devel branch commit prior to: https://github.com/fetchrobotics/fetch_ros/commit/20cd76f4ee2ee1796b36bae3191a1a8b4b959ef7.  Did not revert the change of the estop button's visual file (iirc) or the URDF.

Tested on 18.04 by launching gazebo before and after sourcing a workspace with this branch in it.

Will fix in 20.04 by including this revert commit in #153, but also want to verify visual correctness in 20.04/noetic gazebo.
closed 2021-01-08 by erelson