fetchrobotics/fetch_ros #139: Release request for melodic and kinetic

knorth55, 2020-04-16:
fetch_ros is not released from August 2019.
is it possible to release newer version for melodic and kinetic?

moriarty, 2020-06-11:
Fetch skipped support for Kinetic. I did however release some packages to Kinetic for simulation because there were complaints with the Indigo->Melodic conversion. That was a one-off thing to aid in the transition to melodic and I don't it will continue.

@narora1 or @erelson or @velveteenrobot should be able do a new bloom release.

knorth55, 2020-06-11:
I hope we can get both kinetic and melodic release.
however, we can also wait for kinetic release.

anyway, before melodic release, can you check this PR?
https://github.com/fetchrobotics/fetch_ros/pull/132

knorth55, 2020-11-27:
any update for melodic release of `fetch_ros`?

erelson, 2020-11-28:
While I would like to do a release, and hope it would be painless, it would add a fair bit of testing on our end to verify everything is good. Right now our focus is on working towards a noetic release.  So, I cannot promise anything in this regards for any given timeline.

knorth55, 2020-11-29:
I hope it would be not painful, but I'm relieved that you have a plan for the release.
We are also looking forward to the noetic release, too.