fetchrobotics/fetch_ros #151: [BUG] short description

bergercookie, 2020-10-12:
When I run the fetch gazebo simulation stack and fetch_navigation/fetch_nav.launch (installed from apt) it works fine.
However repeating the same procedure, after cloning fetch_ros (i.e., fetch_navigation) in my local workspace and after compiling the workspace packages with catkin_make, I'm getting the following error when running the navigation (I've the log in this ticket):

```
$ roslaunch fetch_navigation fetch_nav.launch

...

/opt/ros/melodic/lib/move_base/move_base: symbol lookup error: /home/user/ros_ws/devel/lib/libfetch_depth_layer.so: undefined symbol: _ZN2cv4rgbd9depthTo3dERKNS_11_InputArrayES3_RKNS_12_OutputArrayES3_
```

**To Reproduce**

- Clone fetch_gazebo, and fetch_ros in the local workspace, and compile
- roslaunch fetch_gazebo playground.launch on one window
- roslaunch fetch_navigation fetch_nav.launch on the other --> this should fail due to  an undefined symbol

**catkin workspace (please complete the following information):**
 - fetch_ros version: <current melodic-devel> (e831ff0223ca55e8f4f1d6e5ee2ca49fd06ebe7)
 - Ubuntu version: 18.04
 - ROS version: latest melodic from apt

**Additional context**

Running this inside a ubuntu 18.04 docker.

[fetch-undefined-symbol.txt](https://github.com/fetchrobotics/fetch_ros/files/5366434/fetch-undefined-symbol.txt)

moriarty, 2020-10-17:
Have you installed all of the dependencies using rosdep?

When installing via apt, the package dependencies are installed automatically. Building from source in a Docker, will fail to compile without the compile time dependencies, and will fail to run without the execution time dependencies.

http://wiki.ros.org/rosdep#Install_dependency_of_a_particular_package

moriarty, 2020-10-17:
These Dockerfiles are out of date but both run rosdep.

https://github.com/fetchrobotics/fetch_gazebo/issues/46

https://github.com/fetchrobotics/fetch_gazebo/issues/75

bergercookie, 2020-10-23:
>Have you installed all of the dependencies using rosdep?

I hadn't, no, good point.
However, having installed all the dependencies with rosdep, and having removed the `devel/lib/libfetch_depth_layer.so` generated object, I'm still getting this error.

I'm using `catkin_make` instead of `catkin` to compile, but I don't think that difference matters

erelson, 2020-10-23:
This is definitely odd. I haven't taken the time to try the repro (but thanks for the steps!)... a stab in the dark is: You could try putting a CATKIN_IGNORE in the `fetch_depth_layer/` folder and do a clean build from there.

bergercookie, 2020-10-24:
> You could try putting a CATKIN_IGNORE in the fetch_depth_layer/ folder and do a clean build from there.

Sure,  that's actually my current mitigation; it compiles and executes the navigation gazebo demo successfully.