fetchrobotics/fetch_ros #155: [feature] Noetic support?

pabloinigoblasco, 2021-02-12:
Hello.
Is this repo planned to be supported in noetic?

We are using it in our examples in the smacc project and we wanted to move to noetic. We tried to add this project as git submodule and it partially works, but not totally.

https://www.youtube.com/watch?v=FYeNzukdIz4

At the current moment what I did as a workaround is to add the repo as a submodule (forking with [some minor changes ](https://gist.github.com/pabloinigoblasco/851089c0816d07c49963026c44b7e0a3) to at least pass the compilation stage). However it does not work in runtime properly.

erelson, 2021-02-12:
Hi @pabloinigoblasco 

Yep, we essentially have noetic support but haven't released it yet; still need to do more testing on my part, and have been busy the past month or so.  You can check out a few open PRs on this repo and the fetch_robots repo, in particular, with what changes we've got.  Note that for noetic, the branch names we're using are "ros1" instead of "noetic_devel".

I think the equivalent to your linked gist is here: https://github.com/fetchrobotics/fetch_ros/pull/153/files#diff-e121feaf915d1716869198c422dfb66a86b6dfe56f7ebfdb8597cc90b78a39f2R17

One spot my testing ability that is a bit weak is with Gazebo.  Is that part of what your team are experimenting with?

I also haven't personally played with git submodules myself in the past, but feel free to let me know what's not working in that regards.

pabloinigoblasco, 2021-02-15:
I will try that branch you say.

- Yes, we work with gazebo and also the fetch_gazebo package. It was working for melodic but now demos do not work in noetic.

- We are trying to make them work again adding the fetch dependencies as submodules ([github ref](https://github.com/reelrbtx/SMACC/tree/noetic-devel/submodules)). To work in noetic I started forking the melodic branch and introducing some minor changes and I also needed to add some other dependent subrepositories; simple_grasping and robot_controllers.

- I was able to make the demo start in gazebo but the robot did not work, specifically the part of moveit.

I will now retry everything but starting from the ros1 branch.

erelson, 2021-02-21:
> I see in your ros1 branch you still reference 3.2 so maybe that approach you have in that branch is incorrect for noetic ubuntu 20.04.

You might be reading the diffs incorrectly? My link above indicates version is 4.2, same as you're seeing.  Or alternately you're using the "current" ros1 branch, where as my link is to changes that are still in PR.  The erelson/noetic_tweaks branch is currently the latest.  Let me know how it goes, though!

(To be clear, I haven't tried the Gazebo demos, etc. in noetic myself yet)

pabloinigoblasco, 2021-02-26:
Hello.
Thanks for your help. We already have our three fetch demos working based on your erleson/noetic_tweeks pull request. We used the fetch_melodic gazebo version and looks to be working correctly. We have our own gazebo scenarios so we did not tested the gazebo package intensively. 

In our examples we faced three minor issues that may not be related with your work and repositories.

-  the ground truth pluging of gazebo overflows /tf with repeated transforms (that breaks the new policy of tf2 where you cannot repeat transforms with same frame_id and stamp)

- we needed to to some changes in our sdf scene configuration because of some changes on gazebo11 related with the automatic entity naming

- There were also some changes on moveit that affected us (probably not to you), related of the usage of the move_joints functionality (now you cannot send a command to move all the joints of the robot, including head_pan, head_title and fingers if you have selected some specific move_group)

- These are the modules we needed to add for noetic examples (maybe not all are related with fetch examples)

[image]

erelson, 2021-03-28:
Just a quick note: there are ros-testing debians for everything needed for a ros noetic robot now.  I need to test the USB version of the customized 20.04 OS installer still...