fetchrobotics/fetch_ros #165: Can't call mlockall to protect freight_driver and gripper_driver!!!

taylorlulu, 2023-10-29:
We run into this problem when we attempted to start the mechanical arm and gripper of the fetch.
`[ERROR] [1698544305.762663541]: Can't call mlockall to protect freight_driver!!!
[ERROR] [1698544305.762718323]: Can't call nice to prioritize freight_driver!!!

[ERROR] [1698544305.770021604]: initial_covariance is size 0, expected 9.
Using 0.1*Identity.
This probably won't work well.
[ERROR] [1698544305.776856148]: Can't call mlockall to protect gripper_driver!!!
[ERROR] [1698544305.777018399]: Can't call nice to prioritize gripper_driver!!!

[ERROR] [1698544306.053729308]: Couldn't open joystick /dev/fetch_joy. Will retry every second.

[robot_driver-3] process has died [pid 27773, exit code -6, cmd /home/fetch/robot_controllers/devel/lib/fetch_drivers/robot_driver __name:=robot_driver __log:=/home/fetch/.ros/log/b6b8df34-75fd-11ee-a64e-c400ad2eb66e/robot_driver-3.log].
log file: /home/fetch/.ros/log/b6b8df34-75fd-11ee-a64e-c400ad2eb66e/robot_driver-3*.log
[image]

[ WARN] [1698544310.813629074]: imu (IMU) timeout
[ WARN] [1698544310.813740497]: odom (Odometry) timeout
[ WARN] [1698544311.002383345]: Board Not ready: 0x80
[ WARN] [1698544312.002394460]: Board Not ready: 0x80
[ WARN] [1698544313.004358862]: Board Not ready: 0x80
[ WARN] [1698544314.012403438]: Board Not ready: 0x80
[ WARN] [1698544315.012406711]: Board Not ready: 0x80`

As a result the robot can't move mechanical arm and gripper cannot move.

We're using fetch 1084. The version of ubuntu is 18.04. The version of robot_controllers is 0.6.0. The version of fetch_drivers is 0.8.9. We've tried different version of fetch_drivers but it didn't work out.

The rqt_graph
[image]
 is as followed.

erelson, 2023-10-30:
Hi there. I also received an email about this issue, and am addressing the troubleshooting there.

Of note presently is that some of the errors are present on a working robot and can be ignored, namely:
- Can't call mlockall/nice
- initial covariance size
- Couldn't open joystick (will self-resolve)

taylorlulu, 2023-11-01:
I ignored the problem and it worked, thank you @erelson .

closed 2023-11-01 by taylorlulu